TMRh20/AutoAnalogAudio #1: differences with rf24audio?

ricmail85, 2016-11-13:
Hi,
I was looking at this new library. Which are the differences with rf24audio library?

I want to send audio from a nord to the master using the radios. Which library is better?

Thanks!

TMRh20, 2016-11-13:
Hey, thanks for your interest.

The main idea behind this library is to provide some sort of **simplified API/generic interface to the DAC, ADC, Timers and DMA** functionality of Arduino Due that allows a wide range of applications like RF24Audio to be built in short order.  
  
For example, I want to be able to use this to **create other libraries or applications like my RF24Audio or TMRpcm library, or any sort of audio apps**, but for the Arduino Due. Since we are talking Arduino, I also want others to be able to create cool apps without having to learn all about the SAM3X chip & internals.  
  
One main difference is that **AAAudio will not directly include any radio or device specific code**, other than for the Due itself.  From making this, I have a number of ideas that I might look at applying in RF24Audio now also.  

> I want to send audio from a nord to the master using the radios. Which library is better?

Hehe, well that depends. Technically the Due uses 12-bit ADC & DAC where Arduino is 10-bit ADC & PWM. The Due can handle higher rates as well so audio quality can be better. This library can also be made to operate with RF24Audio, acting as a relay & speaker etc. so is much more versatile.

TMRh20, 2016-11-20:
Also just added AVR (Uno, Nano, etc) support.

Am now planning support for many Arduino devices, so that TMRpcm and RF24Audio libs will work on all Arduinos using this as a base.  

In terms of 'better' this will be it.

shubhampaul, 2017-03-01:
btw i ported rf24audio for an external 8-bit dac,didnt do much just removed ramp functions and forwarded the buffer directly to a 8-bit port on uno (atmega328) at timer interrrupt

harmverbeek, 2018-01-21:
I tried WirelessMicrophone and WirelessSpeaker on 2 Arduino Uno.
Didn't compile at first because TcChannel is not defined.
Solved that by putting this in void loop():

`    #if !defined (ARDUINO_ARCH_AVR)
      TcChannel * t = &(TC0->TC_CHANNEL)[0];
      TcChannel * tt= &(TC0->TC_CHANNEL)[1];
    
      Serial.print("Ch0:"); 
      Serial.println(t->TC_RC);      
      Serial.print("Ch1:"); 
      Serial.println(tt->TC_RC);
    #else
      Serial.print("Ch0/1:");
      Serial.println(ICR1);
    #endif
`

And put this line at the top:
`
#define ARDUINO_AVR_ARCH
`

When I play some music on A0 I get this on the Serial Monitor of WirelessMicrophone:
`Ch0/1:996`

And this on the Serial Monitor of WirelessSpeaker:
`Ch0/1:1000` 

But no output.

By the way, RFAudio works on both Unos.

TMRh20, 2018-01-22:
These need to be modified quite a bit to work on AVR. I haven’t gotten around to cleaning up the examples.

harmverbeek, 2018-01-22:
I understand.
The main thing that interested me is that it seems the resolution and speed can be set higher than with rf24audio.

TMRh20, 2018-01-22:
It can handle more because of Arduino Due support. Rf24 audio will provide better sound quality on Arduino Uno etc. for a given sample. I just tested this the other day.

closed 2018-10-17 by TMRh20