TMRh20/AutoAnalogAudio #10: Can wav file hold frequncies in the range >40Khz?

ranibourbara, 2018-12-03:
Hi,

Thank you for this great library.

I am willing to generate a sound wave that is higher than 40Khz. can this be achieved with wav files and this library?

Best,
Rani

TMRh20, 2018-12-03:
Np. Should be able to. I’m not sure of the max frequency, but wav files typically range up to 44khz, so the limiting factor may be the source speed if using an sd card. If using tables in memory, it could possibly go quite faster. Probably over 100khz is not unrealistic.

TMRh20, 2018-12-03:
I should note that is with the Arduino Due only.

ranibourbara, 2018-12-03:
Thank you for your prompt reply. 
So I can generate a wave file with higher sampling rate in Audacity for example. load it onto a SD card (which I see support a clock frequcny in the range of 100Mhz - is this the source speed you are referring to?). I know that the clock speed of the arduino can support this right? so it seems that it might work. I will test this.

TMRh20, 2018-12-03:
Yeah, you can generate files probably around 100khz, 12-bit audio and it should handle it. 
I was referring to the SD read speed being a limiting factor, but with more reading it looks like the Due will top out around 100khz

Peace, 
TMRh20

ranibourbara, 2018-12-03:
TMRh20,

Do you have a proven way to connect an SD or microSD readers to arduino Due? couldn't find my way on the arduino forum. I have disassembeled a a microSD socket from some device and I know it has the pins arranged [3.3, GND, D0, CMD, CLK, D3, D2. D1. C0] but I don't know where to connect these onto the Arduino board.

Thanks

TMRh20, 2018-12-04:
You need to use the ICSP header pins on the Due.

See http://www.thebackshed.com/forum/forum_posts.asp?TID=8341 for pin
connections. ( MISO, MOSI, SCK etc )

See "Connections" under https://www.arduino.cc/en/Reference/SPI for ICSP
pinout etc

TMRh20

ranibourbara, 2018-12-04:
brilliant thank you man

ranibourbara, 2018-12-07:
Hi again,

I have managed to get the SD up and running, and the files uploaded to the Arduino. Trying to play the file that I made I am getting :
SD ok
Analog Audio Begin
Failed to open mySound.wav

Should I change something in the myWav.h file ? or what do you think is the problem. 

Thanks

ranibourbara, 2018-12-07:
Abort. Now working like a charm. Thank you very much

closed 2021-11-06 by TMRh20