TMRh20/AutoAnalogAudio #17: Documentation not updating properly

TMRh20, 2021-04-07:
I'm not sure what exactly the problem is as I'm not too familiar yet with the auto updating of Github Pages from doxygen, so hoping @2bndy5 will come to the rescue.

2bndy5, 2021-04-07:
Docs are only updated on release events. If you want to work around this (without doing a release), you could comment out [this line](https://github.com/TMRh20/AutoAnalogAudio/blob/6e1d7b40592db603789a40f6be3ab8387ef932ed/.github/workflows/doxygen.yml#L42) in _.github/workflows/doxygen.yml_. Beware that every commit on the master branch will update the docs when that line is commented out.

2bndy5, 2021-04-07:
I will also defer to [my latest PR on RF24Audio](https://github.com/nRF24/RF24Audio/pull/36) because that repo is also setup to update the docs on every commit to master, but notice [the doxygen workflow fails](https://github.com/nRF24/RF24Audio/pull/36/checks?check_run_id=2233273738#step:6:222) because my fork has no permission to update upstream/gh-pages branch

2bndy5, 2021-04-07:
ok something is weird with this repo. @TMRh20 could you confirm that the repo settings are using the root directory of the gh-pages branch? It looks like the gh-pages branch is correct.

TMRh20, 2021-04-07:
I just created the gh-pages branch today in attempts to get it working. It is set to that now.

2bndy5, 2021-04-07:
was it only displaying the README before? ok docs are now properly deployed.

I also noticed some problems with the Doxyfile concerning the INPUT and RECURSIVE tags. They should probably be
```
INPUT                  = ./src/AutoAnalogAudio.h \
                         ./src/ESP32/AutoAnalogAudio.cpp
```
and 
```
RECURSIVE              = NO
```

2bndy5, 2021-04-07:
Diving deeper into the documented API, I noticed that the `ramp*()` method weren't getting documented because of the `#if defined ESP32` condition. I've started using a special macro defined by Doxyfile to force inclusion of some platform specific functionality.

Probably needles to say, but I'll submit a PR to address these doc problems

TMRh20, 2021-04-07:
> was it only displaying the README before? 

Yeah, now working again.

PR #18 (2021-04-07): better docs
see [this comment](https://github.com/TMRh20/AutoAnalogAudio/issues/17#issuecomment-815155460) and [this comment](https://github.com/TMRh20/AutoAnalogAudio/issues/17#issuecomment-815198450)

closes #17 

I didn't do the usual main page abstraction nor did I change code fmt to use 4-space indentation.
closed 2021-04-07 by TMRh20