TMRh20/AutoAnalogAudio #20: AutoAnalogAudio

FoxEast2021, 2021-04-17:
Hi Mr
Is AutoAnalogAudio work with esp32 boards?
And if it works, with what kind of esp32 board's?

TMRh20, 2021-04-17:
Any esp32 that uses the Arduino framework and has DAC/ADC pins exposed.

MRcode2102, 2021-04-17:
Can you tell me if this board works with your library please?

https://github.com/Nicholas3388/LuaNode

2bndy5, 2021-04-17:
That link is for the SDK to run a LUA interpreter on a ESP32/8266 based board. It does not demonstrate using the Arduino IDE (which uses c++). Lua is an entirely different programming language than C++.

Please provide a store link to the board you bought.

FoxEast2021, 2021-04-17:
https://www.amazon.co.uk/AZDelivery-NodeMcu-CP2102-Development-including/dp/B071P98VTG/ref=asc_df_B071P98VTG/?tag=googshopuk-21&linkCode=df0&hvadid=430764102847&hvpos=&hvnetw=g&hvrand=13195359199711610181&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=2826&hvtargid=pla-367709801435&psc=1&tag=&ref=&adgrpid=108575012108&hvpone=&hvptwo=&hvadid=430764102847&hvpos=&hvnetw=g&hvrand=13195359199711610181&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=2826&hvtargid=pla-367709801435

at this above link Mr 2bndy5

can you suggest to me an esp 32 board to handling the AutoAnalogAudio library?

2bndy5, 2021-04-17:
It should work with most esp32 boards (including that esp32 dev board) as long as the board has analog input pins (ADC) and analog output pins (DAC). The Arduino framework that TMRh20 mentioned [needs to be installed](https://github.com/espressif/arduino-esp32#installation-instructions) for the Arduino IDE; it should support most esp32 by default.

closed 2021-11-06 by TMRh20