TMRh20/AutoAnalogAudio #21: Useing AAAudio for rec sound's from mic with esp32

FoxEast2021, 2021-04-29:
Hi Mr
I want to use this library to record a wav file with a microphone connected to **esp 32 dev kit**, but my problem is when recording the file because it starts working by pressing the 9 key first, but after pressing the 0 button for a few seconds To stop recording, this message is displayed `File contains no data, exiting
` , and nothing is actually recorded
I used an onboard ADC 34 pin and also had another problem with this library at first and it was in the sd.h library because this library did not match **esp 32** and it showed the error `Failed to open file` that I changed the sd.h library I fixed this error with mySD.h
What do you think is the problem for recording?
please...

TMRh20, 2021-04-29:
There is an issue with the current recording example and ESP32. On Arduino boards like Uno, Nano, Due etc, the SD library behaviour is to append to file, where ESP32 will truncate the file. Essentially, it is probably recording correctly, but the data is  being deleted by ESP32 SD library when the file is opened for finalization. Either that behaviour needs to be modified or the example sketch needs to be modified to fix this issue.

TMRh20, 2021-04-29:
I took a quick look at the recording example, and it looks like a few more modifications are needed to work with ESP32, so I'll see about that...

TMRh20, 2021-04-29:
Ok, I fixed it. The examples can be updated from https://github.com/TMRh20/AutoAnalogAudio/tree/master/examples/SDAudio/SdAudioRecording or if you reinstall the library manually it will work as well.

FoxEast2021, 2021-04-29:
Thank you very much **Mr. TMRh20** for this great favor
Just a little help
If you can give me a hint about the ADC ESP32 PIN because now I put the ADC PIN like this `  aaAudio.begin(39, 1);     // Start AAAudio with ADC & DAC
`  like as this [ESP32 DEV KIT V1 PINOUT'S](https://acoptex.com/uploads/ESP32Espresiffboards12.jpg) the WAV file is recorded correctly on the SDCARD but there is no sound inside the recorded WAV file
Do you think the problem of choosing the wrong ADC PIN is?
THANKS FOR YOUR REPLYS

FoxEast2021, 2021-04-29:
Thank you very much Mr. TMRh20 for this great favor

FoxEast2021, 2021-04-29:
Thank you very much for your answers
Now it records the wav file on the sdcard, but my only problem is the sound is recorded because the sound inside the wav file is very fast, which means that we have doubled the playback speed of the wav file while playing it!
This is the case. Do you think there is a problem with my esp32?
If you want, I can upload a sample of the recorded file for you, of course, if needed, sir!

TMRh20, 2021-04-29:
Hmm, I'm not sure why it would be speeding up on playback like that. It could have something to do with the SD write speed. What if you try a lower sample rate. ( Call ```startRecording(newWavFile, 11000);``` and ``` stopRecording(newWavFile, 11000); ``` )

It should also show the ```Samples per Second:``` while recording so you can see if there are buffer underruns.

FoxEast2021, 2021-05-01:
thank you very much (Great Sir)

FoxEast2021, 2021-05-01:
Excuse me, Mr. TMRh, for my many questions from you
I wanted to use this [library ](https://github.com/hjltu/esp8266-wifi-microphone)and [your ](https://github.com/TMRh20/AutoAnalogAudio)library to send the recorded audio to the mqtt server through the mqtt server I set up on my computer, and then save it using node-red, but a small problem I have about how to configure node-red; To store audio sent from esp32.
I wanted you to give me a reference or guidance on this if possible. (Ie for example the node-red side, should a buffer be created to receive it?)
Thanks for your great answers

TMRh20, 2023-04-26:
Sorry about the late reply, but looks like you're on your own on this one anyway as I don't know the answer. I can't really help much more with this issue.

closed 2023-04-26 by TMRh20