TMRh20/AutoAnalogAudio #22: Can we get support esp8266 board?

krishnakumar4a4, 2021-05-23:
I am trying to use a microphone with my NodeMCU V2 board and stream audio PCM data to a service over WIFI. I am using PlatformIO IDE to program this board. Last time I tried using this, it gave an error(undefined reference to functions) while compiling to ESP8266 target. I am not sure it is because of lack of support for ESP8266 board or I am missing some compiler options. Could someone help? @TMRh20

TMRh20, 2021-05-23:
Esp8266 is not supported. :(

closed 2021-11-06 by TMRh20

raphox, 2023-01-26:
Hi @TMRh20 
Could you give us more info about the related limitations? For example, is it a software or hardware limitation?
I am not an expert, but maybe I could spend some time trying to propose something to make it possible.

TMRh20, 2023-01-26:
AFAIK the limitations are the device itself/OS. There are a number of reasons nobody has made a lib for it but I’m not really the expert on 8266 to get into details