TMRh20/AutoAnalogAudio #29: Which code Example can i use to record sound with  the Arduino Due

JoelMuhanguzi, 2022-07-15:
Hello @TMRh20 I have made a connection to the arduino due using a max4466 mic as well as an sd card, tried using the sdaudiorecording library but it keeps popping an error failed to start recording, could you help with an example that I could use with the arduino due to be able to acquire an audio sample from the mic, thank you

closed 2022-07-15 by JoelMuhanguzi

TMRh20, 2022-07-15:
Hoping you closed this because you got it working! The examples should work out of the box with the Due, as the library was initially designed for the Due.

JoelMuhanguzi, 2022-07-15:
yes please, thanks very much, i have been able to capture sound from the microphone and store it on an sd card, thats why i closed the issue, it is really nice to learn that it was initially designed for the Due.