TMRh20/AutoAnalogAudio #32: SD compability

killamorri07, 2022-12-06:
Hi, I'm using Arduino Due.
My goal is to write and save an audio with 8bit and sr = 16kHz to sd card. I'm using SdFat32. I have test run all component by itself however when I tried to combine SdFat32 and modify the code from your example, I can not initiate SdFat32 at all. It always gave me an error(can not find SdFat32). 

Thank you in advance. I really looking forward to hear from you.

TMRh20, 2022-12-06:
What is the exact error message you are getting?
Also can you post some example code that demonstrates the issue?

killamorri07, 2022-12-07:
I have not written my own code, I just change the SD.h part to SdFat and file to file32. 

`/******** User Config ************************************/

#define sd_CS_PIN 4
#define AUDIO_DEBUG
#define RECORD_DEBUG

const char newWavFile[] = "test.wav";

/*********************************************************/
#include <SPI.h>
#include <SdFat.h>
#include <sdios.h>
#include <AutoAnalogAudio.h>
/*********************************************************/
AutoAnalog aaAudio;
SdFat32 sd;
File32 myFile;
File32 recFile;
/*********************************************************/
#include "myWAV.h"
#include "myRecording.h"
/*********************************************************/

void setup() {

  Serial.begin(115200);

  if (!sd.begin(sd_CS_PIN)) {
    Serial.println("sd init failed!");
    return;
  }
  Serial.println("sd ok\nAnalog Audio Begin");

  aaAudio.begin(1, 1);     // Start AAAudio with ADC & DAC
  aaAudio.autoAdjust = 0;  // Disable automatic timer adjustment

}

/*********************************************************/
uint32_t displayTimer = 0;

void loop() {

  if (millis() - displayTimer > 1000) {
    displayTimer = millis();
    if (counter) {
      Serial.print("Samples per Second: ");
      Serial.println(counter * MAX_BUFFER_SIZE);
    }
    counter = 0;
  }

  if (Serial.available()) {
    char input = Serial.read();
    switch (input) {

      case '1':  playAudio("M8b24kM.wav");  break; //Play a *.wav file by name - 8bit, 24khz, Mono
      case '2':  playAudio("M8b24kS.wav");  break; //Play  8bit, 24khz, Stereo
      case '3':  playAudio("M16b24kS.wav"); break; //Play 16bit, 24khz, Stereo
      case '4':  playAudio("M8b44kST.wav"); break; //Play  8bit, 44khz, Stereo
      case '5':  channelSelection = 0;      break; //Play the audio on DAC0
      case '6':  channelSelection = 1;      break; //Play the audio on DAC1
      case '7':  channelSelection = 2;      break; //Play the audio on DAC0 & DAC1
      case '8':  Serial.println("OK");      break;
      case '9':  startRecording(newWavFile, 11000); break; //Start recording @11khz,8-bit,Mono
      case '0':  stopRecording(newWavFile, 11000);  break; //Stop the recording and finalize the file
      case 'p':  playAudio(newWavFile);      break; //Play back the recorded audio
      case 'D':  sd.remove(newWavFile);      break; //Delete the file and start fresh
[... 242 lines cut ...]
#if defined (RECORD_DEBUG)
    Serial.println("OK");
  } else {
    Serial.println("Failed");
#endif
  }
  recFile.close();

}

/*********************************************************/

void stopRecording(const char *fileName, uint32_t sampleRate) {

  aaAudio.adcInterrupts(false);                        //Disable the ADC interrupt
  recFile.close();                                         //Close the file
  createWavHeader(fileName, sampleRate);                   //Add appropriate header info, to make it a valid *.wav file
#if defined (RECORD_DEBUG)
  Serial.println("Recording Stopped");
#endif
}
`

I am not sure is it because I did not have `#define SPI_CLOCK SD_SCK_MHZ(21)` or I just simpliy misunderstood your code. 
[image]

killamorri07, 2022-12-07:
okay, I will try... thank you :D

closed 2022-12-07 by killamorri07