TMRh20/AutoAnalogAudio #33: AutoAnalogAudio.h conflicts with Audio.h library

alcarazolabs, 2022-12-13:
Hi dear, I'm having an issue with the `AutoAnalogAudio.h` , I'm using esp32 board. The problem happens
when I include in my project the `Audio.h` library, I upload my code to the board no problems but when
I use the Audio.h library the library throws this message:

```
Analog Audio Begin
E (37) I2S: /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2s.c:1252 (i2s_adc_enable):i2s built-in adc not enabled
```
It says `"i2s built-in adc not enabled"` I don't know if there is a way to enable adc, I'm not esp32 expert, it is for that reason that I'm writing this issue to ask for help.

The library `AutoAnalogAudio.h` works well when I record audio, however when I include the Audio.h library it throws that message and I'm no unable to record audio because the board starts to reboot.

Here is code:

```
// esp32 board v.1.0.6
#include <SPI.h>
#include <SD.h>
//=================================================
#include <AutoAnalogAudio.h>

#define SD_CS_PIN 5
#define AUDIO_DEBUG
#define RECORD_DEBUG

AutoAnalog aaAudio;
File myFile;
File recFile;
#include "myWAV.h"
#include "myRecording.h"

const char newWavFile[] = "/record.wav";

//=================================================

//SD:
#define SD_MISO     19
#define SD_MOSI     23
#define SD_SCLK     18
#define SD_CS       5

//Amplifier MAX98357 PINS:
#include "Audio.h"
Audio audio;

#define I2S_DOUT      25 //DIN PIN
#define I2S_BCLK      27
#define I2S_LRC       26

void setup() {
  Serial.begin(115200);
  pinMode(SD_CS, OUTPUT);
  if (!SD.begin(SD_CS)) {
    Serial.println("SD init failed!");
    return;
  }
  Serial.println("SD ok\nAnalog Audio Begin");
  
  
  // AutoAnalogAudio.h
  aaAudio.begin(1, 1);     // Start AAAudio with ADC & DAC
  aaAudio.autoAdjust = 0;  // Disable automatic timer adjustment

  //set Audio.h
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(10); // 0...21

}

uint32_t displayTimer = 0;
bool recording = false;

void loop() {
   //"Audio.h"
  // audio.loop();

   
  //AutoAnalogAudio.h
  if (millis() - displayTimer > 1000) {
    displayTimer = millis();
    if (counter) {
      Serial.print("Samples per Second: ");
      Serial.println(counter * MAX_BUFFER_SIZE);
    }
    counter = 0;
  }

  if (Serial.available()) {
    char input = Serial.read();
    switch (input) {
      case 'r':  startRecording(newWavFile, 24000); recording = true; break; //Start recording @11khz,8-bit,Mono
      case 's':  stopRecording(newWavFile, 24000);  recording = false; break; //Stop the recording and finalize the file
      case 'p':  playRecordedAudio(newWavFile);      break; //Play back the recorded audio
      case 'D':  SD.remove(newWavFile);      break; //Delete the file and start fresh
    }
  }

#if defined (ESP32)
  if(recording){
    ADC_Handler();
  }
#endif

}

void playRecordedAudio(const char *file){
//   audio.setVolume(10); // 0...21
  // audio.connecttoFS(SD, file);
   
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreaminfo(const char *info){
    Serial.print("streaminfo  ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}
void audio_eof_speech(const char *info){
    Serial.print("eof_speech  ");Serial.println(info);
}

I will appreciate any idea to fix this problem.
thanks so much.
```

The Audio.h library is from here:
ESP32-audioI2S https://github.com/schreibfaul1/ESP32-audioI2S

thanks.

TMRh20, 2022-12-13:
I don't think these libraries are compatible. The problem is most likely that both libraries are trying to use the I2S peripheral. 
Unfortunately there is probably not an easy solution.

alcarazolabs, 2022-12-13:
> I don't think these libraries are compatible. The problem is most likely that both libraries are trying to use the I2S peripheral. Unfortunately there is probably not an easy solution.

Thanks for the reply. I want record an audio and then play it, its possible by using AutoAnalogAudio.h library? I saw an example here https://tmrh20.github.io/AutoAnalogAudio/SdAudioWavPlayer_8ino-example.html that it inside the switch condition it call a method called playAudio() I tested this method but I can't listen it in my speaker, I'm using a MAX98357 amplifier where this are my connected pins:

#define I2S_DOUT      25 //DIN PIN
#define I2S_BCLK      27
#define I2S_LRC       26
As you can see dear, my idea is use a library which allows me to record and play this audio files.
I tried Audio.h because this can play a remote .wav file and with AutoAnalogAudio it can't.

I will spect the answer from the author of Audio.h library and see if I can enable I2S. Now I ask you, how I can play audio by using a MAX98357 amplifier with AutoAnalogAudio.h 

thanks in advance.

TMRh20, 2022-12-14:
I developed AAAudio on the board shown below, which uses pins 25 and 26 for I2S output.

[image]

I would suggest looking at the included ESP32 example and stealing code from there to make it playback audio.
Please note that ESP32 support is still a bit buggy, so you may find better luck if you can get the Audio library working.

alcarazolabs, 2022-12-14:
> I developed AAAudio on the board shown below, which uses pins 25 and 26 for I2S output.
>
> [image]
>
> I would suggest looking at the included ESP32 example and stealing code from there to make it [...]

Thanks man I will do.

closed 2022-12-15 by alcarazolabs