TMRh20/AutoAnalogAudio #34: Problem with ADC read

alazka8, 2023-02-22:
Hi,  I used your lib and have some problem with adc :)
I used it with adcChannel = ADC1_CHANNEL_5; and I need get value from analogRead(36).
When program comes to line with analogread it freeze.
I try adc1_get_raw(ADC1_CHANNEL_0) and still the same.
Is there any way to use lib and read value from ADC1_CHANNEL_0 ?

TMRh20, 2023-02-23:
AutoAnalogAudio does its own thing with the ADC, so if you are trying to use it separately that may cause problems. I don't have a solution for you, but there is probably a way to manually disable the ADC functionality of AAAudio. It depends on what board you are using.

alazka8, 2023-02-23:
I need record from ADC1 channel5 and get value from ADC1 channel0 at the same time. I use esp32 wrover

alazka8, 2023-02-23:
> AutoAnalogAudio does its own thing with the ADC, so if you are trying to use it separately that may cause problems. I don't have a solution for you, but there is probably a way to manually disable the ADC functionality of AAAudio. It depends on what board you are using.

I have esp32 wrover, I need record from adc1_channel_5 and read value from adc1_channel_0 at the same time ;//

alazka8, 2023-03-06:
> I have esp32 wrover, I need record from adc1_channel_5 and read value from adc1_channel_0 at the same time ;//

:) Can you tell me how to disable it manually ?

TMRh20, 2023-04-26:
Sorry for the late reply, must have missed this one. I don't know the answer, and ESP32 support is still pretty buggy/in development.

closed 2023-04-26 by TMRh20