TMRh20/AutoAnalogAudio #37: Recording speed

Geethukulal, 2023-05-18:
Recording speed is too fast is there is  any solution for that

TMRh20, 2023-05-18:
What board are u using?Typically you can just adjust the sample rate to fine tune playback speed

Geethukulal, 2023-05-18:
Iam using ESP32 Board,  if i record 10 sec audio in memory card it will show 2 min of recording

TMRh20, 2023-05-18:
I hate to say it, but ESP32 support is very buggy right now. I should probably remove it for now, but in any case, I need to do a bunch of the work on the library to solve your problem.

Geethukulal, 2023-05-18:
Okey,

Geethukulal, 2023-05-18:
Can i use max9814 microphone to record audio

TMRh20, 2023-05-18:
Yeah that's what I typically use

Geethukulal, 2023-05-18:
How can i connect it

Geethukulal, 2023-05-19:
For which pin of esp32 board  i connect out pin of max9814

TMRh20, 2023-05-19:
It depends on the board, but whichever pin is A0 or Analog pin 0 in the Arduino platform

closed 2023-05-19 by Geethukulal