TMRh20/AutoAnalogAudio #39: Undetected type when using Arduino IDE

notevanrutten, 2023-05-26:
```
c:\Users\*****\OneDrive\Documents\Arduino\libraries\AutoAnalogAudio\src/AutoAnalogAudio.h:258:3: error: 'adc1_channel_t' does not name a type; did you mean 'i2s_channel_t'?
   adc1_channel_t adcChannel;
   ^~~~~~~~~~~~~~
   i2s_channel_t
```

I'm using the Arduino IDE to try to record audio using the XIAO Sense and an external electret microphone.  I followed the setup outlined in the [documentation](https://wiki.seeedstudio.com/xiao_esp32s3_getting_started/) before uploading any code. I'm wondering if there's a missing dependency that wasn't installed initially or if it's a more underlying issue.

TMRh20, 2023-05-26:
It looks like it is not detecting your board correctly. You need to make sure to select the Xiao Ble sense board in Arduino IDE and it only works with the mbed-enabled core.

TMRh20, 2023-05-26:
My mistake, i thought u were using the xiao ble sense not esp32. Just fyi esp32 support is very buggy. I’ll do some checking to see if your board would be supported however

TMRh20, 2023-05-26:
Unfortunately it looks like your board is not supported by this library :(

closed 2023-05-26 by TMRh20