TMRh20/AutoAnalogAudio #42: RP2040 solution

pierrotm777, 2024-12-28:
Hello,
I use your email address for the TMRpcm lib on a Pro Micro and love it.
But I should like now use a RP2040 and I don't find a solution like your.
Can you accept please to add RP2040 to your lib?
Thanks for your help,
Pierre

TMRh20, 2024-12-28:
I will give it some thought, I have a RPi Pico but it would still take some time.

pierrotm777, 2024-12-28:
No problem, I can wait 😀.
Thanks

pierrotm777, 2024-12-28:
If that can help you, I have tried with success on my RP2040 the PWMAudio example which run very well but, need a header file. I should like as tmrpcm does it, load a file from sd and play with.

pierrotm777, 2025-01-02:
I have found a nice lib for RP2040, [BackgroundAudio](https://github.com/earlephilhower/BackgroundAudio) witch does what I should like but without possibility to change speed an pitch sound.
I send you my demo witch run very well.
May be you could add your own code or create a fork of this lib ?
[NotSoSimpleWAVShuffle.zip](https://github.com/user-attachments/files/18293701/NotSoSimpleWAVShuffle.zip)