TMRh20/AutoAnalogAudio #43: Issue with example SDrecording

rickyelqasem, 2025-01-14:
I'm using an ESP32 and I'm going directly off the pins (no I2s components) 

I'm using the default settings:

aaAudio.begin(1, 1);     // Start AAAudio with ADC & DAC
aaAudio.autoAdjust = 0;  // Disable automatic timer adjustment

but in the serial output I get this.. 

SD ok
Analog Audio Begin
E (17) i2s(legacy): i2s_check_cfg_validity(930): I2S buffer length at most 1024 and more than 8
E (17) i2s(legacy): i2s_driver_install(1582): I2S configuration is invalid
E (23) i2s(legacy): i2s_adc_enable(898): Not initialized yet
E (28) i2s(legacy): i2s_set_clk(1042): I2S0 has not installed yet

Is there anything I have not setup or is it an issue?

rickyelqasem, 2025-01-15:
So i noticed a bit more of the error when I try to perform any action:

Start Recording: /test.wav
E (8427) i2s(legacy): i2s_set_clk(1042): I2S0 has not installed yet
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400da93c  PS      : 0x00060130  A0      : 0x800d4278  A1      : 0x3ffb9660  
A2      : 0x00000000  A3      : 0x3ffcad94  A4      : 0x00003200  A5      : 0x3ffb96ac  
A6      : 0x00000000  A7      : 0x3ffce460  A8      : 0x3ffce460  A9      : 0x3ffb9660  
A10     : 0x00000001  A11     : 0x3ffce7c8  A12     : 0xffffffff  A13     : 0x3ffcea68  
A14     : 0x000020ec  A15     : 0x00000004  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000014  LBEG    : 0x400847a1  LEND    : 0x400847a9  LCOUNT  : 0x00000027  

Backtrace: 0x400da939:0x3ffb9660 0x400d4275:0x3ffb96a0 0x4008c0b6:0x3ffb96d0

ELF file SHA256: 5496005f8d5b4629

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15488
load:0x40080400,len:4
load:0x40080404,len:3180
entry 0x400805b8
SD ok
Analog Audio Begin

TMRh20, 2025-01-15:
ESP32 support never worked quite properly, and it looks like I broke it more on the last update.

I might just remove the ESP32 support altogether unless I get a chance to fix it.

rickyelqasem, 2025-01-15:
If I switch to a different MCU would you say this works nicely on an Arduino Zero, which has 1 DAC?

TMRh20, 2025-01-15:
It was designed on Arduino Due, so works very well on that platform. I don't think it will work with the Zero.

I would recommend the nRF52840 boards like Feather Express 52840, Clue or XIAO Sense 52840. It is super versatile, supporting **Input**: PDM, SAADC or I2S or **Output**: PWM, I2S
It also has an onboard radio which can support bluetooth or operate using ESB for live streaming of audio etc. using the https://github.com/TMRh20/nrf_to_nrf library.

rickyelqasem, 2025-01-17:
So in your example in your blog for the nRF52840:

aaAudio.dinPin = 1; 
aaAudio.begin(3, 1);
sets up output on AIN0, which is A4, but how do you define the record pin?

rickyelqasem, 2025-01-17:
So these are the defaults just to make sure I get it the right way around ?

[image]

TMRh20, 2025-01-17:
Sorry, its all confusing because I was confused when writing the code for nRF52 at first.

The PWM pins are defined by GPIO numbers, so when I say pin 5 port 0 I mean the pin marked P0.05 or A0 in other words. The only way to change it is to edit the AutoAnalogAudio.cpp file and modify `DEFAULT_PWM_PIN` and `DEFAULT_PWM_PORT` settings. The pin with your red arrow would be P1.08 (GPIO pin 8, port 1)

I don't really recommend using PWM though since it will be low quality compared to I2S.

rickyelqasem, 2025-01-17:
Thanks I was a bit confused..

closed 2026-02-14 by TMRh20